MotorolaMobilityLLC/kernel-mtk #2: Defconfig for Taido_row

HostZero, 2016-12-21:
I have this kernel source: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-XT1662-S0529-160920

i have gone trough commit history their is taido_row_debub_defconfig

but in kernel source i couldnt find kernel config.

which config i should use?

- HostZero

jenish97, 2017-01-14:
Found defconfig
https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/MMI-XT1706-S131-160905/arch/arm/configs

HostZero, 2017-01-14:
Okay! So they are two different defconfigs.. Like
Mine is for XT1662. Is their any different defconfig. As the tag you have provided is of XT1706.

jenish97, 2017-01-14:
Hey on link for taido defconfig file available for xt1706

HostZero, 2017-01-15:
Thankyou! I will try a compile with that config.

closed 2017-01-15 by HostZero

jenish97, 2017-01-15:
Any other which file miss
And can give your contact no. For what's app

And turjya shah port miui 8 camera and nav bar not work
My patch to volte work but volte symbol not show