MotorolaMobilityLLC/kernel-mtk #4: Code compile

mmehdi, 2017-05-06:
Hi guys
I am trying to compile the code on Ubuntu.
I have setup CROSS_COMPILE and ARCH variables. When i run
make -j5 ARCH=arm, I get the following error

`make: *** [tools/dct/DrvGen] Error 2`

please help?

thanks

chenyt9, 2017-05-08:
What is the code tag? Perhaps you can get compiled instruction information from this URL:  https://github.com/MotorolaMobilityLLC/readme
Thanks.

mmehdi, 2017-05-09:
ok I have managed to compile and generate the Kernel image.
is there any guide to unpack stock Boot image and port the custom Kernel?

thanks

turjyasaha, 2017-05-12:
Download the Stock Firmware.
You'll get boot.img file into the stock firmware folder.
You can Unpack the boot image by any third-party software.

mmehdi, 2017-05-12:
@turjyasaha I am unable to extract/unpack the boot.img in my stock rom. It's a working rom, I can flash it on my Moto E but can't unpack the boot.img. I have tried many third party tools
If you can have a look at my boot.img please here https://www.dropbox.com/s/ugenfduao339a0h/boot.img?dl=0

thank you

mmehdi, 2017-05-12:
@turjyasaha I am unable to unpack my boot.img from the stock rom. It's a working rom, i have tested it on my phone but cannot unpack the boot.img. I have tried many third party tools.
Can you have a look at my boot.img please?
https://www.dropbox.com/s/ugenfduao339a0h/boot.img?dl=0

thanks

turjyasaha, 2017-05-13:
Dear @mmehdi,
Above boot.img file is fully encrypted.
 You can't unpack the file. But
You can download this Firmware
https://tinyurl.com/m688sh6
And it is  easy to unpack  "boot.img" by any third-party software.
Mediatek Extractor- unpack img file
https://tinyurl.com/mhsyyl5
We're building LineageOS for this device and solving some bugs.
If you need to help, you can contact me.
Happy to help you.

mmehdi, 2017-05-14:
@turjyasaha Hi yes I would like to contact you, please give me your email and I will send you what I am trying to do with Android Kernel.

Thanks

turjyasaha, 2017-05-15:
@mmehdi 
Email id:  turjyasaha@gmail.com

mmehdi, 2017-05-15:
@turjyasaha I have sent you an email. Thank you

closed 2017-11-01 by chenyt9