MotorolaMobilityLLC/kernel-mtk #7: Moto C kernel source

Mysteryagr, 2017-09-09:
I need Moto C kernel source please

chenyt9, 2017-10-09:
Hi , @Mysteryagr 

We have a process to follow before to release a version to Github. We are working to release XT1754  code, it will be available in a few days.

Thanks,

Mysteryagr, 2017-10-09:
The same model as my device, great news!

Thanks @chenyt9

darshan1205, 2017-10-10:
@chenyt9 XT1721

Mysteryagr, 2017-10-30:
Any news for XT1754 please? @chenyt9

zhangheting407051, 2017-11-01:
Dear  all:
Moto C Plus kernel source has been released
Thanks!

zhangheting407051, 2017-11-01:
https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/nougat-7.1.1-release-lacrosse

Mysteryagr, 2017-11-01:
What about Moto C only, namath? @zhangheting407051

Zormax, 2017-11-01:
@zhangheting407051, ha-ha-ha
this source https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/nougat-7.1.1-release-lacrosse
is fake for Lacrosse (XT1723)
this source - reference source mediatek
not A158_defconfig or lacrosse_defconfig , dts, lcm, and all other code for Moto C Plus

Mysteryagr, 2017-11-01:
@Zormax why do you think so?
This source can be build with gr6737m defconfig as it appears, but I don't have Moto C plus to make sure that the defconfig has the right drivers.

Mysteryagr, 2017-11-01:
Anyway I opened this issue for Moto C 4G (namath) XT1754 kernel source, so please open another issue if want to discuss C plus stuff :D

Zormax, 2017-11-01:
this is fake source. dot.

Mysteryagr, 2017-11-01:
@Zormax any proof? Why Motorola developers would push fake sources?
Download this app and compare your device drivers with the drivers in the defconfig
https://play.google.com/store/apps/details?id=ru.andr7e.deviceinfohw
If everything seems fine, compile, flash and test.

Zormax, 2017-11-01:
Pfff. you are very naive and a little know about the code.

Mysteryagr, 2017-11-01:
@Zormax sounds like you are talking about yourself, you can't even prove what you are saying.

Zormax, 2017-11-01:
Good luck in the compilation, ha-ha-ha

Mysteryagr, 2017-11-01:
@Zormax kernel for my device isn't released it, good luck to you too :)

chenyt9, 2017-11-02:
@Mysteryagr ，
I'm very sorry. Responsible for the XT1754 code release is another colleague, he is still less familiar with the release of the process, I have tried to urge him to release the code. I have seen namath own the highest priority, I believe that our colleagues will soon release the code. But I think he should take a little time. Please give us a little time, thank you!
BRs.
Yuting

zhangheting407051, 2017-11-02:
@Mysteryagr @Zormax @chenyt9 
Very sorry, this is my mistake, I have updated the Moto C Plus source.
The source: https: //github.com/MotorolaMobilityLLC/kernel-mtk/tree/nougat-7.1.1-release-lacrosse
Thank you

Mysteryagr, 2017-11-02:
@chenyt9
No problem, I am waiting

WTspm, 2017-11-10:
Dear all:
Moto C 4G kernel source has been released
Thanks!

Mysteryagr, 2017-11-10:
I have a small question before I close this issue, which defconfig I should build with?
I can see there are many defconfigs in this source:

> wt6737m_35_n_debug_defconfig
> wt6737m_35_n_defconfig
> wt6737m_35g_n_debug_defconfig
> wt6737m_35g_n_defconfig
> wt6737t_36_n_debug_defconfig
> wt6737t_36_n_defconfig
> wt6737t_36g_n_debug_defconfig
> wt6737t_36g_n_defconfig

Thanks in advance
@WTspm @chenyt9  @zhangheting407051  @changxiaoqiang1

Zormax, 2017-11-10:
@Mysteryagr, Developers do not put the right defconfig

Mysteryagr, 2017-11-10:
@Zormax how are we supposed to build without the right defconfig? :D

Zormax, 2017-11-10:
no way

Mysteryagr, 2017-11-10:
@Zormax, if it is so, and the source is the right one, only the defconfig is missing, we can port a defconfig, I ported different MTK kernel sources - different versions - before from other device to mine: https://github.com/Mysteryagr/Mystery-Kernel-3.10
Booted with many bugs.

But why the struggle!? can't we have a complete kernel source to work with!!!

I need an answer from the developers.

Zormax, 2017-11-11:
They have not invested at least: defconfig, dts, codegen
E. We cut all the most valuable

Mysteryagr, 2017-11-11:
@Zormax Got it

Ahhh, please push full sources! @changxiaoqiang1 !!!

WTspm, 2017-11-13:
Hi all，
The defconfig file is existed，you can choose defconfigs as follow:
wt6737m_35_n_debug_defconfig
wt6737m_35_n_defconfig
thks!

Mysteryagr, 2017-11-13:
Okay thanks, will compile and test @WTspm

Mysteryagr, 2017-11-13:
Great source code!!!

`Error: ./arch/arm/boot/dts/cust.dtsi:19.2-3 label or path, 'nfc', not found
Error: arch/arm/boot/dts/wt6737m_35_n.dts:668.2-3 label or path, 'nfc', not found
FATAL ERROR: Syntax error parsing input tree
make[1]: *** [arch/arm/boot/dts/wt6737m_35_n.dtb] Error 1
make: *** [dtbs] Error 2
make: *** Waiting for unfinished jobs....`

`In file included from drivers/mmc/host/mediatek/mt6735/mt_sd_misc.c:36:0:
./drivers/misc/mediatek/include/mt-plat/sd_misc.h:26:19: fatal error: mt_sd.h: No such file or directory
compilation `terminated.`

`In file included from drivers/misc/mediatek/base/power/mt6735/mt_pbm.c:38:0:
./drivers/misc/mediatek/include/mt-plat/mt6735/include/mach/mt_thermal.h:29:24: fatal error: mt_gpufreq.h: No such file or directory
compilation terminated.
make[6]: *** [drivers/misc/mediatek/base/power/mt6735/mt_pbm.o] Error 1
make[6]: *** Waiting for unfinished jobs....`

Why you don't upload a fully working source without missing files and errors!?

Zormax, 2017-11-13:
@Mysteryagr 
wt6737m_35_n_defconfig * - is the standard (the reference) file from the mediatek (wingtech)
And what did they tell you about it - they have deceived you

Mysteryagr, 2017-11-13:
I believed you now

Zormax, 2017-11-13:
I repeat once again: Original defconfig, dts, codegen - no
No need to keep me for a fool

Mysteryagr, 2017-11-14:
I could bypass those compiling errors but I don't want to waste my time on an incomplete kernel.

Again to the developers, please provide a real source code.

chenyt9, 2017-11-14:
@Mysteryagr @Zormax 
Sorry, there may be misunderstandings here, I need to figure out the situation. Now which project is currently compiled unsuccessful? Both Moto C Plus (lacrosse) and Moto C 4G (namath)? Or only Moto C 4G (namath)? I will communicate with the owner who posted the code. Believe me, Motorola is very sincere, we have some mistakes in the release of the code, but we are trying to understand the problem and try to solve.
Thanks.
chenyt9

Zormax, 2017-11-14:
@chenyt9, for namath

Mysteryagr, 2017-11-14:
I have moto C 4G namath XT1754, that's the device I am working on.

WTspm, 2017-11-15:
Dear all:
Moto C 4G namath kernel source is released, please check.
Thanks!

Mysteryagr, 2017-11-15:
I am going to build and test

Mysteryagr, 2017-11-15:
The kernel still has missing files, are we supposed to fix that by ourselves or what!?

In file included from drivers/misc/mediatek/base/power/mt6735/mt_pbm.c:38:0:
./drivers/misc/mediatek/include/mt-plat/mt6735/include/mach/mt_thermal.h:29:24: fatal error: mt_gpufreq.h: No such file or directory
compilation terminated.
make[6]: *** [drivers/misc/mediatek/base/power/mt6735/mt_pbm.o] Error 1
make[5]: *** [drivers/misc/mediatek/base/power/mt6735] Error 2
make[4]: *** [drivers/misc/mediatek/base/power] Error 2
make[3]: *** [drivers/misc/mediatek/base] Error 2
make[2]: *** [drivers/misc/mediatek] Error 2
make[1]: *** [drivers/misc] Error 2
make: *** [drivers] Error 2
make: *** Waiting for unfinished jobs....

Mysteryagr, 2017-11-15:
I know how to add the missing header file and fix the fatal error, what I am referring to that the Kernel is still incomplete; shows compiling errors.

Mysteryagr, 2017-11-15:
And again ...
have you tested this source before pushing it to github!?

`./drivers/misc/mediatek/btcvsd/inc/mt6735/AudDrv_BTCVSD.h:18:28: fatal error: AudioBTCVSDDef.h: No such file or directory
compilation terminated.
make[4]: *** [drivers/misc/mediatek/btcvsd/AudDrv_BTCVSD.o] Error 1
make[3]: *** [drivers/misc/mediatek/btcvsd] Error 2
make[3]: *** Waiting for unfinished jobs....`

WTspm, 2017-11-16:
Dear ,
I'm sure the source code is complete, and I compile the code successful.
I see you feedfack the error, the file is missed, you can check the source you download .
And on the github, I noticed you can find the file you missed, please check.
Or you can tell me the compile instructions.
Thks.

Mysteryagr, 2017-11-17:
@WTspm
I fixed it here: https://github.com/Mysteryagr/Moto-C-namath-XT1754-kernel/commit/00af20f1f0d8a9a0bd4c637ab3d7a826f3597b8b
Now the compiler can see the file, I think it wasn't placed in the right directory, but editing ccflags did the trick.
Anyway, another compiler error has poped up, I guess I will have to fix them one by one now.

Mysteryagr, 2017-11-18:
Okay.. so actually I don't build on my PC, I build using travis, what I did is importing your source and make travis compile it with linaro 5.2 toolchain, my point is the probability of wrong download can't be assumed, still the source have so many missing files, and the compiler doesn't see other, I had to edit some ccflags, and everytime I fix / bypass a compiling error another one appears, so if any one could compile and check I will be thankful, just want to make sure it is not my mistake, because you are saying you compiled the source before pushing it :)

Mysteryagr, 2017-11-20:
Please update the kernel

Mysteryagr, 2017-11-23:
After so many attempts I came up with the following:
- Your guide for namath in readme repo doesn't work
- Your toolchain for namath doesn't work

What I did is downgrading my toolchain from linaro 5.2 to linaro 4.9 and the kernel compiles successfully, still didn't flash and test it.

So, feel free to close this issue

Kind regards

closed 2017-11-25 by Mysteryagr