MotorolaMobilityLLC/kernel-mtk #10: XT1754/XT1723

Zormax, 2017-10-07:
Please put source code for XT1754/XT1723

darshan1205, 2017-10-07:
@chenyt9

chenyt9, 2017-10-09:
Hi , @darshan1205 @Zormax  

We have a process to follow before to release a version to Github. We are working to release the code, it will be available in a few days.

Thanks,

Zormax, 2017-10-09:
@chenyt9 for XT1771 MMI-NMA26.42-97 ???

chenyt9, 2017-10-10:
Hi, @Zormax
Version MMI-NMA26.42-97 is released.
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-NMA26.42-97

Zormax, 2017-10-20:
Hi, @chenyt9 
Why do you collect models XT1754 / XT1723 lite kernel (lk) with the option MTK_SEC_FASTBOOT_UNLOCK_SUPPORT = no
???
Even companies like Meizu, Sony, NOKIA has decided to appeal for products that open the possibility of release.
Here, a series of budget phones and you have collected lite kernel with option MTK_SEC_FASTBOOT_UNLOCK_SUPPORT = no
Very bad. On the E-Series and E + - good developers, gathered with the option MTK_SEC_FASTBOOT_UNLOCK_SUPPORT = yes

Zormax, 2017-10-31:
@chenyt9, any news for source XT1754/XT1723 ???

Zormax, 2017-11-01:
@chenyt9, ha-ha-ha
this source https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/MMI-ALPS-MP-N0.MP1-V1.0.2
is fake for Lacrosse (XT1723)
this source - reference source mediatek
not A158_defconfig or lacrosse_defconfig , dts, lcm, and all other code for Moto C Plus

WTspm, 2017-11-10:
Dear all:
XT1754 kernel source has been released
Thanks!

Zormax, 2017-11-11:
@WTspm 
They have not invested at least: defconfig, dts, codegen
E. We cut all the most valuable

WTspm, 2017-11-15:
Dear all:
XT1754 source has been released, please check.
Thanks!

closed 2018-05-28 by Zormax