MotorolaMobilityLLC/kernel-mtk #14: Camera hals broken

PainKiller3, 2017-12-19:
Please fix the camera hals as fast as possible the camera show blank

PainKiller3, 2018-03-08:
@chenyt9
please have a look...
kernel source for moto M (XT1663) is incomplete..
camera stuff is missing...

PainKiller3, 2018-03-08:
Doesn't work still same : ( @chenyt9

chenyt9, 2018-03-09:
@PainKiller3，please confirm that you are using the correct tag version: MMI-XT1663-S356-170621.
Thanks.

PainKiller3, 2018-03-10:
@chenyt9 yes i m using this tag https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/MMI-XT1663-S356-170621

chenyt9, 2018-03-13:
@PainKiller3, camera stuff in uder the path:drivers/misc/mediatek/imgsensor. Can you provide your compile command? Thanks.

chenyt9, 2018-03-15:
@PainKiller3 , I fetch the code and try to compile and find a file missing two definitions:
The error message is:
'SENSOR_DRVNAME_S5K3M2_SUNNY_MIPI_RAW' undeclared here (not in a function)
'SENSOR_DRVNAME_OV8865_SUNNY_MIPI_RAW' undeclared here (not in a function)
If you have the same error message as me, please add the following two lines to the file "drivers/misc/mediatek/imgsensor/inc":
#define  SENSOR_DRVNAME_OV8865_SUNNY_MIPI_RAW         "ov8865sunnymipiraw "
#define SENSOR_DRVNAME_S5K3M2_SUNNY_MIPI_RAW          "s5k3m2sunnymipiraw "
If your error message is not the same, please send your build command and build log to me. I will update the readme file and update the two lines of code to the source file. If you still have questions, please contact me.

chenyt9, 2018-03-15:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-XT1663-S356-170621.txt

Pipaolo, 2018-03-16:
@chenyt9 I tried adding that, it compiled successfully. But when i flashed it to test. The camera won't work for some reason. :/

PainKiller3, 2018-03-17:
@chenyt9  the camera is not working even after following the steps given in the text file : ( pls fix the code : ( thanks

chenyt9, 2018-03-22:
@PainKiller3  @Pipaolo , maybe it's because of a missing macro control. Please add this definition when compiling: TARGET_PRODUCT := kungfu_m。
Please try it. If you still have problems, please let me know. I would like to know how you guys perform the brushing and please send me your compiler instructions and flashing steps. I try to find another reason from elsewhere. Thanks.

Pipaolo, 2018-03-22:
@chenyt9 I tried adding that an error happened "make: *** empty variable name.  Stop."

chenyt9, 2018-03-22:
@Pipaolo , are you compiled according to the steps of this document ( https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-XT1663-S356-170621.txt ) ? I did not encounter the situation as you said, I added "TARGET_PRODUCT := kungfu_m" in the makefile, or you directly export TARGET_PRODUCT=kungfu_m.
Or, you can send me your build command.

Pipaolo, 2018-03-22:
@chenyt9 where do I add "TARGET_PRODUCT := kungfu_m"?  I literally added that in the make command.

chenyt9, 2018-03-22:
@Pipaolo , I think it is also possible, but I directly export this variable, or choose to add in the Android.mk file in the root directory of the kernel...I think there are many ways, as long as it can be read.  
Because we didn't compile the camera code before, we use this variable to control the camera's compilation，You can look at the makefile below these paths, search for this variable, and you will understand:
Kernel-3.18/drivers/misc/mediatek/flashlight/src/mt6755
Kernel-3.18/drivers/misc/mediatek/imgsensor/src/mt6755
Kernel-3.18/drivers/misc/mediatek/imgsensor/src/mt6755/camera_hw
Thanks.

closed 2018-04-26 by chenyt9