MotorolaMobilityLLC/kernel-mtk #18: Failure to release build able kernel

onejay09, 2018-01-23:
The panelli branch doesn't build
Net/Netfilter fails to compile

DhirajSurvase, 2018-01-26:
i added missing stuff and headers in netfilter
compiled successfully....
getting black screen after boot........

DhirajSurvase, 2018-01-26:
will compile and post adb logs for crash...

onejay09, 2018-01-26:
I saw your work, they are supposed to release build able source, even after I fixed the errors it can't load past bootanimation

DhirajSurvase, 2018-01-30:
@chenyt9
any news......
given kernel source code is incomplete and buggy...
sir have a look pleaz..

DhirajSurvase, 2018-01-31:
[panelli.txt](https://github.com/MotorolaMobilityLLC/kernel-mtk/files/1680825/panelli.txt)
here are the adb logs

DhirajSurvase, 2018-01-31:
[dmesg.log](https://github.com/MotorolaMobilityLLC/kernel-mtk/files/1680844/dmesg.log)

here are dmesg logs

DhirajSurvase, 2018-01-31:
@zhangheting407051 sir.......
reply....we r waiting....

zhangheting407051, 2018-01-31:
@DhirajSurvase 
Excuse me, now is the compilation failed?

DhirajSurvase, 2018-01-31:
noo..
i added missing netfiler stuff from namnath..
and fixed permissions..
still not booting.

DhirajSurvase, 2018-01-31:
https://github.com/DhirajSurvase/android_kernel_moto_panelli/blob/master/build.sh

DhirajSurvase, 2018-01-31:
it's xt1721 moto c plus

zhangheting407051, 2018-02-01:
@DhirajSurvase

zhangheting407051, 2018-02-01:
@DhirajSurvase 
https://github.com/MotorolaMobilityLLC/kernel-mtk/blob/nougat-7.1.1-release-panelli/readme.txt
Please use the compile command in readme.txt
Thank you

onejay09, 2018-02-01:
> @DhirajSurvase <https://github.com/dhirajsurvase>
> https://github.com/MotorolaMobilityLLC/kernel-
> mtk/blob/nougat-7.1.1-release-panelli/readme.txt
> Please use the compile command in readme.txt
> Thank you

We tried it there's parts of the source code missing

DhirajSurvase, 2018-02-01:
@zhangheting407051
Sir I tried ...
First it was giving permission errores..
After fixing permissions it was giving missing files and headers errores...
Some files are missing...
After adding those stuff now kernel not booting...

DhirajSurvase, 2018-02-01:
https://github.com/DhirajSurvase/kernel-mtk/commits/nougat-7.1.1-release-panelli
1.../tools/dct/DrvGen.py :permission denied
2. tools/dct/old_dct/DrvGen: Permission denied
3.missing files from net/netfilter
after this i compiled it but it's not passing bootanimation.....
@zhangheting407051

zhangheting407051, 2018-02-02:
OK, sir
I am working on it and I expect to solve it tomorrow
thank you for your feedback

zhangheting407051, 2018-02-03:
@DhirajSurvase 
https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/nougat-7.1.1-release-panelli
The code has been modified and updated, please follow the steps in readme.txt to compile and verify
Thank you!

DhirajSurvase, 2018-02-05:
@zhangheting407051 
now it compiled successfuly...
is something missing or wrong because kernel getting bootloop ..
is it tested on device after updating??

zhangheting407051, 2018-02-05:
@DhirajSurvase 
Yes, please update the latest source code before testing
Thank you

DhirajSurvase, 2018-02-05:
@zhangheting407051 
Yes I used updated stuff...
Same bootloop issue...
From my side everything is fine.....

zhangheting407051, 2018-02-07:
@DhirajSurvase 
I'm sorry, I do not understand what you mean.
Is there any problem now?

onejay09, 2018-02-07:
@zhangheting407051

The kernel fails to boot, it doesn't work.

zhangheting407051, 2018-02-08:
@onejay09 
Hello, can you describe your steps in detail, I will follow your steps to try
Thank you

DhirajSurvase, 2018-02-08:
@zhangheting407051 
cd kernel-3.18
mkdir -p out
make O=out ARCH=arm A158_debug_defconfig
make -j32 O=out ARCH=arm CROSS_COMPILE=/home/AD/moto/Kpanelli/prebuilts/gcc/linux-x86/arm/arm-eabi-4.8/bin/arm-eabi-

then flashed zImage-dtb from out/arch/arm/boot/
as i did with my other 3 devices..
it was expected to boot...
flashed on both stock and custom roms.....
it stucks on bootanimation when i use debug defconfig..
i tried A158_defconfig...it booted but again shutdown.....
again flashing old kernel it boots fine...
Request you to just test these sources on device once...

zhangheting407051, 2018-02-08:
@DhirajSurvase 
Okay, I think I got it, I follow your steps to check it
Thank you

zhangheting407051, 2018-02-08:
@DhirajSurvase 
The code has been modified and updated,Please help confirm!
Thanks

DhirajSurvase, 2018-02-08:
@zhangheting407051 
thanks
sir booted fine now...

@onejay09 close issue now

closed 2018-02-08 by onejay09