MotorolaMobilityLLC/kernel-mtk #20: Moto C Plus wifi bugs (panelli)

nandanmenon, 2018-03-21:
I am facing a lot of issues with wifi. It used to disconnect automatically. A restart was required to fix that.
Please fix the bug.

DhirajSurvase, 2018-03-22:
@zhangheting407051 @chenyt9
have a look..
even bluetooth also turns on automitically..
and signal strength also very low..
many users crying on https://forums.lenovo.com
and the moto's agents are so noob on that forum giving silly answers to the users on that forum...

DhirajSurvase, 2018-03-22:
also the charging is damm slow it takes more than 5 hours with 2A charger on your stock kernel..

DhirajSurvase, 2018-03-22:
https://forums.lenovo.com/t5/Moto-C-Moto-C-Plus/Moto-c-plus-slow-charging-issue/td-p/4002226

https://forums.lenovo.com/t5/Moto-C-Moto-C-Plus/Flashlight-and-camera-are-not-in-sync/td-p/4003173

https://forums.lenovo.com/t5/Moto-C-Moto-C-Plus/Jio-4G-VoLTE-NETWORK-BLINKS-FREQUENTLY/td-p/3855840

here are some noob answers given by agents...

awheeler-mmi, 2018-03-22:
@nandanmenon and @DhirajSurvase, thanks for the feedback!

We will the comments back to the internal teams.  

Also, if you have kernel patches for fixes then we can help shepherd them to the kernel developers.

In the meantime, check back on Lenovo support threads for updates.

DhirajSurvase, 2018-03-23:
except kernel source i can't find any branch for panelli in other repos..
looks there is development going on for this device ..

DhirajSurvase, 2018-03-24:
i am using some wood's blobs  ..
i don't have any issue like wifi and bt

zhangheting407051, 2018-03-26:
@DhirajSurvase 
What are the operating steps? Is there any special operation?
Please help confirm the software version in the phone? Have you upgraded to the latest software?

nandanmenon, 2018-03-26:
yep i it is the latest version

closed 2018-03-26 by nandanmenon

DhirajSurvase, 2018-03-27:
Why you closed issue @nandanmenon....
Ofc I am on latest firmware. .