MotorolaMobilityLLC/kernel-mtk #28: am I right : "Moto is ignoring Low Budget devices"??

DhirajSurvase, 2018-07-02:


Zormax, 2018-07-03:
Here is another example:
Moto E4 - 113-148-152 latest firmware
Moto E4plus - only 113 firmware with OTA

chenyt9, 2018-07-10:
@all，
If you have a version requirement, please post the project name and version number. Thanks.

chenyt9, 2018-07-10:
@Zormax ,
The lastest release is  updated : 
Moto E4 : https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-NMA26.42-146
Moto E4plus: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-NMA26.42-152

I don't quite understand what you mean, could you explain  it in more detail? Do you still have new version requirements?？
Thanks.

chenyt9, 2018-07-10:
@zhangheting407051

Zormax, 2018-07-11:
@chenyt9 no problem

DhirajSurvase, 2018-08-24:
Rip Moto...

closed 2018-08-24 by DhirajSurvase