MotorolaMobilityLLC/kernel-mtk #40: KERNEL REQUEST

Karl621, 2021-01-08:
Good Afternoon
Please release kernel source for POA29.550_132_2010_20201027 (for Motorola Moto e6 play XT2029-1) I require it for compile an alternative custom mod (Since December 20th I'm having problems with the current version POAS29.550_132_2_2012_20201212) I called to the customer service for warranty but the person said that it was "invalid" only for my IMEI (supposed, it is a foreign smartphone) when I bought it on my country via amazon (October 8th 2020). After I recovery using "Rescue and Smart Assistant" and keep installing the same defective version, I sent multiple request for remove this version but nothing change. I need a working smartphone.
Thanks.

mbissaromoto, 2021-01-08:
Hi @Karl621 

Transfered from kernel-msm (that is for qualcomm based phones) to here, as Moto E6 Play (bali) is Mediatek phone.

+ @chenyt9

Karl621, 2021-01-09:
> Hi @Karl621
> 
> Transfered from kernel-msm (that is for qualcomm based phones) to here, as Moto E6 Play (bali) is Mediatek phone.
> 
> * @chenyt9

Ok, thanks (but trying post the original request on MTK devices only appeared a comparition between files page.)

ontimoss, 2021-01-19:
The workflow has been triggered and is expected to be completed within 3 weeks. Please be patient, thank you.

Karl621, 2021-01-19:
> The workflow has been triggered and is expected to be completed within 3 weeks. Please be patient, thank you.

Thank you very much, I will be patient and stay pendent during the next three weeks.

ontimoss, 2021-03-18:
https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/MMI-POAS29.475-62-26-7

 Update finish.

closed 2021-03-18 by ontimoss

Karl621, 2021-03-18:
> https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/MMI-POAS29.475-62-26-7
> 
> Update finish.

Thank you very much.