MotorolaMobilityLLC/kernel-mtk #42: custom private key .pem moto e6 play

GKart, 2021-04-14:
I don't know if this should be here or where it should be but I need the private key to be able to sign my custom vbmeta
i have a moto e6 play bali model xt2029-1 poas29.550-132-3 RETAR and i want to create a custom vbmeta to be able to use custom software

GKart, 2021-04-14:
or some method to disable boot verification

100Daisy, 2022-06-20:
You can disable AVB or sign system,boot,vendor,vbmeta using custom avb key. 
Signing kernel is different thing. If you want to use .ko drivers sign key must be same for kernel binary and drivers. Sign keys should be confidential, to prevent loading untrusted modules so motorola won't share them. Use custom key and push both drivers and kernel binaries to the device.

closed 2022-07-26 by awheeler-mmi