MotorolaMobilityLLC/kernel-mtk #46: Build issues with pie-9.0.0-release-poker branch

Vasant-1234, 2021-08-08:
I was trying to build for my Motorola E6 Plus phone and build fails in
kernel-mtk/tinno/common_features/touch/focaltech_touch_screen/.
It is missing focaltech_config.h among other things. Please fix build.

blankting, 2021-08-11:
@Vasant-1234 Did you choose a right defconfig, such as "p161m_defconfig"?

blankting, 2021-08-11:
> I was trying to build for my Motorola E6 Plus phone and build fails in
> kernel-mtk/tinno/common_features/touch/focaltech_touch_screen/.
> It is missing focaltech_config.h among other things. Please fix build.

@Vasant-1234 Did you choose a right defconfig, such as "p161m_defconfig"?

Vasant-1234, 2021-08-11:
I used the config that I extracted from my Moto E6 Plus phone (/proc/config.gz).

blankting, 2021-08-13:
@Vasant-1234 The code contains the "focaltech_config.h" you mentioned，see 
"tinno/p160an/touchpanel/focaltech_touch_screen/focaltech_config.h"

Please refer to follow steps:
my_top_dir =$PWD
Download prebuilts folder from Android distribution and checkout corresponding branch. 
For example:
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9

cd arm-linux-androideabi-4.9

git checkout pie-release

cd $my_top_dir

cross=$my_top_dir/arm-linux-androideabi-4.9/bin/arm-linux-androideabi-

kernel_out_dir=$my_top_dir/out

cd kernel-mtk

make ARCH=arm CROSS_COMPILE=$cross O=$kernel_out_dir p161m_defconfig

make ARCH=arm CROSS_COMPILE=$cross O=$kernel_out_dir KCFLAGS=-mno-android headers_install

TARGET_PRODUCT=p161m make ARCH=arm CROSS_COMPILE=$cross O=$kernel_out_dir KCFLAGS=-mno-android -j8

KatoTempest, 2022-04-09:
this commands really work on the arm64 variant?
i try to adapt these instructions for my variant but recive various errors

`~/android/kernel-mtk$ bash build.sh
Makefile:422: KBUILD_CFLAGS -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89
drivers/../tinno/common_features/dev_info/Makefile:5: ---[WD]---
cat: /home/licht/android/kernel-mtk/build.ini: No existe el archivo o el directorio
cat: /home/licht/android/kernel-mtk/build.ini: No existe el archivo o el directorio
drivers/../tinno/common_features/fingerprint/Makefile:11: ---------------fingerprint_supprot----------------
cat: /home/licht/android/kernel-mtk/build.ini: No existe el archivo o el directorio
.
.
.
.
PLATFORM = 
DEVAPC platform code exists...
************  drivers/trusty/mtee-kree mk ************
build trng...
TCORE_UT_FWK_SUPPORT = n
TCORE_PROFILING_SUPPORT = n
TCORE_PROFILING_AUTO_DUMP = n
PMEM_MOCK_OBJECT_SUPPORT = n
TCORE_MEMORY_LEAK_DETECTION_SUPPORT = n
Makefile:422: KBUILD_CFLAGS -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89
drivers/../tinno/common_features/dev_info/Makefile:5: ---[WD]---
cat: /home/licht/android/kernel-mtk/build.ini: No existe el archivo o el directorio
cat: /home/licht/android/kernel-mtk/build.ini: No existe el archivo o el directorio
drivers/../tinno/common_features/fingerprint/Makefile:11: ---------------fingerprint_supprot----------------
cat: /home/licht/android/kernel-mtk/build.ini: No existe el archivo o el directorio
.
.
.
.
PLATFORM = 
DEVAPC platform code exists...
************  drivers/trusty/mtee-kree mk ************
build trng...
TCORE_UT_FWK_SUPPORT = n
TCORE_PROFILING_SUPPORT = n
TCORE_PROFILING_AUTO_DUMP = n
PMEM_MOCK_OBJECT_SUPPORT = n
TCORE_MEMORY_LEAK_DETECTION_SUPPORT = n
make[1]: se entra en el directorio '/home/licht/out'
/home/licht/android/kernel-mtk/Makefile:422: KBUILD_CFLAGS -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
arch/arm64/configs/p161bn_defconfig:482:warning: override: reassigning to symbol MTK_PWM
#
# configuration written to .config
#
make[1]: se sale del directorio '/home/licht/out'
make[1]: se entra en el directorio '/home/licht/out'
/home/licht/android/kernel-mtk/Makefile:422: KBUILD_CFLAGS -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89
  CHK     include/generated/uapi/linux/version.h
  HOSTCC  scripts/unifdef
  INSTALL usr/include/asm-generic/ (35 files)
  INSTALL usr/include/drm/ (21 files)
  INSTALL usr/include/linux/android/ (1 file)
  INSTALL usr/include/linux/byteorder/ (2 files)
  INSTALL usr/include/linux/caif/ (2 files)
  INSTALL usr/include/linux/can/ (5 files)
  INSTALL usr/include/linux/dvb/ (8 files)
  INSTALL usr/include/linux/hdlc/ (1 file)
  INSTALL usr/include/linux/hsi/ (2 files)
  INSTALL usr/include/linux/iio/ (2 files)
  INSTALL usr/include/linux/isdn/ (1 file)
[... 963 lines cut ...]
  CC      fs/proc/kmsg.o
  LD      fs/sysfs/built-in.o
  CC      fs/proc/page.o
  CC      fs/aio.o
  CC      fs/locks.o
make[1]: *** [/home/licht/android/kernel-mtk/Makefile:1112: kernel] Error 2
  CC      fs/sdcardfs/derived_perm.o
  CC      fs/compat.o
  CC      fs/compat_ioctl.o
  LD      fs/proc/proc.o
  LD      fs/proc/built-in.o
  CC      fs/binfmt_script.o
  CC      fs/binfmt_elf.o
  CC      fs/compat_binfmt_elf.o
  CC      fs/mbcache.o
  CC      fs/posix_acl.o
  LD      fs/sdcardfs/sdcardfs.o
  LD      fs/sdcardfs/built-in.o
  CC      fs/coredump.o
  CC      fs/drop_caches.o
  CC      fs/dcookies.o
  LD      fs/built-in.o
make[1]: se sale del directorio '/home/licht/out'
make: *** [Makefile:152: sub-make] Error 2
`

ItsIgnacioPortal, 2023-08-16:
Does anyone know the difference between these two kernels?
- "MMI-PTA29.401-25"
- "MMI-PTB29.401-25"

They're both for the `poker` branch, but which one should be used for the arm64 variant?

asbraga-motorola, 2023-09-08:
I'm closing this issue as it is too old and did not inform the specific build id. Please open a new one if still necessary.

closed 2023-09-08 by asbraga-motorola

ItsIgnacioPortal, 2023-09-09:
How would one get the specific build id?