MotorolaMobilityLLC/kernel-mtk #49: GPL | Source request for Motorola Moto Edge 20 Lite (Kyoto) RRK31.Q3-19-97

Hasaber8, 2022-01-14:
CC: @chenyt9

chenyt9, 2022-01-14:
Hi，@Hasaber8 ，
I have triggered the release process of the version, it may take at least two weeks, thank you in advance for your patience.

Hasaber8, 2022-01-14:
Thank you @chenyt9

Hasaber8, 2022-02-16:
@chenyt9 @awheeler-mmi it's been over a month now.

forforksake, 2022-02-19:
@chenyt9 can you please share your progress on this
Many thanks

ElytrA8, 2022-02-20:
Well I wanted to ask is fusion variant also known as Kyoto ?

Hasaber8, 2022-02-20:
> Well I wanted to ask is fusion variant also known as Kyoto ?

Yep its the same.

forforksake, 2022-02-20:
Just dimensity 800 for fusion and 720 for lite

ElytrA8, 2022-02-20:
So do we have sources yet, i.e
Kernel Trees
Device Trees
Vendor Trees

Hasaber8, 2022-02-20:
Fyi vendor and device trees are proprietary.
Only kernel tree is released.

ElytrA8, 2022-02-20:
So we can start developing initial oss device Trees, right ?

ElytrA8, 2022-02-20:
Is someone working on this

ElytrA8, 2022-02-20:
@Hasaber8 sir plz answer if you can

forforksake, 2022-02-20:
I've started looking at it.
Just gathering what I can from the boot.img
Will take another look later hopefully

chenyt9, 2022-02-24:
Hi，all，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRK31.Q3-19-97
Thank you all for your patient！

closed 2022-02-24 by chenyt9