MotorolaMobilityLLC/kernel-mtk #57: [Request] Moto G50 5G Kernel Sources

alienatiz, 2022-05-10:
I want to improve Moto G50 5G (codenamed as "saipan"). 
Can I receive the kernel source data of Moto G50 5G?

(branch needed as "android-11-release-rrs")

alienatiz, 2022-05-10:
> 

Android version 11, and its codename is saipan.

alienatiz, 2022-05-10:
> Hello @alienatiz,
> 
> Could you please specify the version you are looking for? Thank you.

Android version 11, and its codename is saipan.

mbissaromoto, 2022-05-10:
Hi @alienatiz 

We need to known the exact released version you want. Example: RRS31.QX-YY-ZZ

alienatiz, 2022-05-11:
> Hi @alienatiz
> 
> We need to known the exact released version you want. Example: RRS31.QX-YY-ZZ

Here it is, RRSS31.Q2-54-59-6

chenyt9, 2022-05-11:
Hi, @alienatiz , will arrange the release for RRSS31.Q2-54-59-6.

SantiiAgoo5121, 2022-05-12:
> Hi, @alienatiz . Can you help me? I need Moto G20 Kernel Sources RTA31.68-29

chenyt9, 2022-06-15:
Hi, @alienatiz ,
Version released:https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/RRSS31.Q2-54-59-6
Thanks.

alienatiz, 2022-06-15:
> Hi, @alienatiz , Version released:https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/RRSS31.Q2-54-59-6 Thanks.

Thanks, I've checked it.