MotorolaMobilityLLC/kernel-mtk #69: Request Kernel Sources TONGA RQQS31.Q3.68.149-2

Ekylypse, 2022-10-26:
Request Kernel Sources TONGA RQQS31.Q3.68.149-2 using MediaTek Helio G37 (MT6765H)

chenyt9, 2022-12-21:
@Ekylypse ,
This version RRQS31.Q3.68.149-2 is already on our release list, thank you very much for your patience. We will publish it as soon as possible.
Here is two release for Moto g power ：
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRQS31.Q3-68-140-2
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRQ31.Q3-68-16-2R1
Can you please use those version first？Thanks a lot.

chenyt9, 2023-01-19:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRQS31.Q3-68-149-2

closed 2023-01-19 by chenyt9