MotorolaMobilityLLC/kernel-mtk #71: Kernel Sources for Moto G72 (Victoria) MTK G99

AbzRaider, 2022-11-19:
Please release the latest linux kernel source for the device asap.

asbraga-motorola, 2022-12-01:
Hi @AbzRaider 
Can you inform the exact released version you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2.

thenoelboi, 2022-12-04:
S3SV32.45-24-3 seems to be the latest available for MOTO G72

uname -r output:  5.10.110-android12-9-00005-g4fc5e12e2daf-ab8748924

AbzRaider, 2022-12-28:
@asbraga-motorola when release ?

atulnakul121, 2023-01-10:
@asbraga-motorola @thenoelboi release when?
Waiting from November.

atulnakul121, 2023-01-16:
Sir please release it soon.
Atleast please reply.

atulnakul121, 2023-01-16:
> Please release the latest linux kernel source for the device asap.

Release soon @asbraga-motorola

chenyt9, 2023-01-18:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-S3SV32.45-24-3
Thank you all for your patience.

closed 2023-01-18 by chenyt9