MotorolaMobilityLLC/kernel-mtk #79: [Request] Moto e6s (fiji_64) Kernel sources

partedMagic, 2023-05-07:
Hello, could you guys please release kernel source for the e6s 4/64 version, fiji_64?

asbraga-motorola, 2023-05-12:
Can you inform the exact released version you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2.

partedMagic, 2023-05-19:
> Can you inform the exact released version you want? You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2.

by any chance, is it this? POES29.288-60-6-1-29
or maybe this? PQ2A.190405.003

partedMagic, 2023-06-16:
when 💀

asbraga-motorola, 2023-10-09:
Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-POES29.288-60-6-1-29