MotorolaMobilityLLC/kernel-mtk #80: GPL | Source request for Motorola Edge 40 (XT2303) (lyriq) G_TTL33.51

FuccDucc, 2023-05-12:
Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola Edge 40 (lyriq) version `G_TTL33.51`

Model numbers: `XT2303-1`, `XT2303-2`, `XT2303-2-DS`, `XT2303-2-SS`
Codename: `lyriq`

TAG=G_TTL33.51

* It's mediatek.. i put in the wrong repository first, now i move it from msm to mtk

chenyt9, 2023-05-26:
@FuccDucc ,
for Motorola edge 40, we only shipped a versionT2TL33.3-15-4
This version of TTL33.51 is no found. Pls make a double confirm.
For T2TL33.3-15-4, you can refer to:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-T2TL33.3-15-4

FuccDucc, 2023-05-27:
Correction: T2TL33M.3-22-5

> @FuccDucc , for Motorola edge 40, we only shipped a version:T2TL33.3-15-4 This version of TTL33.51 is no found. Pls, make a double confirm

@chenyt9 Correction: T2TL33M.3-22-5 is what i want.

CloudedQuartz, 2023-07-04:
For the record, update T2TL33.3-22-9 has also been pushed yet source code has not been released

CloudedQuartz, 2023-07-06:
LYRIQ_G_T2TL33.3-49-2 update has been pushed also, yet source has not been released

chenyt9, 2023-07-14:
T2TL33.3-22-5 version is released:

https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-T2TL33.3-22-5

chenyt9, 2023-07-27:
@CloudedQuartz 
Version released:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-T2TL33.3-22-9

closed 2023-07-27 by chenyt9

chenyt9, 2023-07-27:
@CloudedQuartz 
MMI-T2TL33.3-49-2 version is here：
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-T2TL33.3-49-2