MotorolaMobilityLLC/kernel-mtk #81: [Request] Kernel sources for Motorola Edge (2022) codename tesla

mfossen, 2023-05-24:
Hello! Please release kernel sources for Motorola Edge (2022) codename `tesla`.

Build number: S2STS32.71-118-4-2

Thanks!

rangersix7, 2023-05-24:
Hey mfossen. Me too...Please release kernel source for Motorola Edge 2022 XT2205 code name Tesla.
I don't think it is Motorola not releasing the source code as much as it is MediaTek. MediaTek just does enough to keep GPL happy. We can hope and ask..

chenyt9, 2023-05-26:
@all, pls use this verion first:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-S2ST32.71-89
The open source process of version S2STS32.71-118-4-2 is already in our workflow.
Please be patient. Thanks.

closed 2023-05-26 by chenyt9

mfossen, 2023-05-26:
Thank you @chenyt9!

rangersix7, 2023-05-26:
Thank you chenyt9