MotorolaMobilityLLC/kernel-mtk #82: [REQUEST] MILAN_RETUS_12_S3RDE32.123-37

PizzaG, 2023-05-25:
Hello, I would like to request Kernel Source and Vendor Blobs needed for: 
XT2211-2_MILAN_RETUS_12_S3RDE32.123-37_subsidy-DEFAULT_regulatory-DEFAULT_cid50_R1_CFC

Thanks in advance

PizzaG, 2023-06-13:
I think an oopsie was made on this?  This is 4.14 kernel and what's on the device reports as 4.19.191+    Thanks in advance

electimon, 2023-06-13:
@asbraga-motorola

PizzaG, 2023-07-07:
> Fixed: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-S3RDE32.123-37-1
> 
> Please note that I deleted the old tag and created a new one with the same name.
> 
> This is the right commit for the tag:
> 
> ```
> git rev-list -n 1 MMI-S3RDE32.123-37-1
> 27e458c4c3f9263a2025c9fa7251baf68d25d696
> ```
> 
> If you get a different value, you should delete the local tag and fetch it again:
> 
> ```
> git tag -d MMI-S3RDE32.123-37-1
> git fetch origin tag MMI-S3RDE32.123-37-1
> ```

Awesome, than you much for getting it fixed, very much appreciated!!  Have a good one =)