MotorolaMobilityLLC/kernel-mtk #84: [Request] Moto G Power 2022, Tonga | S3RQS32.20-42-10-9-2

electimon, 2023-06-06:
Kindly, and thanks.

asbraga-motorola, 2023-06-24:
Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-S3RQS32.20-42-10-9-2

electimon, 2023-06-25:
@asbraga-motorola tonga is 4.19 on android 12, the uploaded source is 4.14. kindly correct, thank you.

electimon, 2023-06-30:
@asbraga-motorola @chenyt9 bump

asbraga-motorola, 2023-07-05:
Fixed: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-S3RQS32.20-42-10-9-2

Please note that I deleted the old tag and created a new one with the same name.

This is the right commit for the tag:
```
git rev-list -n 1 MMI-S3RQS32.20-42-10-9-2
ffd0ffe3c0bc9d5ceb2a94526f771d559a17f42f
```

If you get a different value, you should delete the local tag and fetch it again:
```
git tag -d MMI-S3RQS32.20-42-10-9-2
git fetch origin tag MMI-S3RQS32.20-42-10-9-2
```

electimon, 2023-07-05:
Thank you so much, it indeed looks like the right kernel this time. Besides this issue there is also #82 which has the wrong kernel as well. Please when you can take a look, ill close this issue now.