MotorolaMobilityLLC/kernel-mtk #87: kernel sources, drivers, firmware and vendor blobs of moto e-13

arunkumar413, 2023-08-15:
Hi Team,

Could you please release the kernel source, drivers, firmware and vendor blobs of moto e-13 (https://www.motorola.in/smartphones-moto-e13/p?skuId=318)

Thanks,
Arun

asbraga-motorola, 2023-08-15:
Hi @arunkumar413 
Can you inform the exact released version (Build ID) you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2.

arunkumar413, 2023-08-17:
@asbraga-motorola I didn't buy the mobile yet. I plan to buy it once I have the necessary source code and blobs. Does the source code vary from one mobile to another despite being of the same model?

asbraga-motorola, 2023-08-17:
Yes, we need the Build ID for releasing the exact version.

arunkumar413, 2023-08-20:
This requirement is for porting capyloon https://capyloon.org/ to moto e-13 . It requires kernel sources and device trees for porting. Do you also release the device tree as well? If not please let me know the process to fetch device tree as well

arunkumar413, 2023-08-22:
> This requirement is for porting capyloon https://capyloon.org/ to moto e-13 . It requires kernel sources and device trees for porting. Do you also release the device tree as well? If not please let me know the process to fetch device tree as well

@asbraga-motorola Could you please look into this

arunkumar413, 2023-08-28:
@asbraga-motorola Here is the build id: TLA33. 105-226-2 

Not sure if there should be a space after 33. Attaching the image for your reference.

[image]

arunkumar413, 2023-09-04:
@asbraga-motorola Any update on this?

arunkumar413, 2023-09-11:
@asbraga-motorola Please let me know if this will take some more time.

asbraga-motorola, 2023-09-12:
It's in the process to be released, may take some time.

arunkumar413, 2023-10-04:
@asbraga-motorola Just wanted to check if there is any progress on this request.

asbraga-motorola, 2023-10-16:
The code has been published in the kernel-sprd repo:
https://github.com/MotorolaMobilityLLC/kernel-sprd/tree/MMI-TLA33.105-226-2

closed 2023-10-16 by asbraga-motorola