MotorolaMobilityLLC/kernel-mtk #89: Kernel sources request for Moto G Stylus 2023 (non-5G) XT2317-2 (gnevan)

Giantvince1, 2023-09-09:
About Phone > Built Number: T1TH33.75-12-6-1-3
Output of uname -r: 4.19.191-g7d9f49e9ff52

I am requesting these sources so that one may develop a custom ROM or recovery image for this device. It has a great feature set for a budget phone and is also bootloader unlockable, and custom ROM support would be an amazing way to keep this phone going for many years.

Giantvince1, 2023-09-15:
Any update on this? I'm really hoping to start up a custom scene for this device.

Giantvince1, 2023-09-28:
How might I get some eyes on this request for device trees for this device? I cannot start development without them, and allowing this device the ability to get custom roms would be a huge favor to the modding community at large- especially for those who own retail US versions of this device with unlocked bootloaders.

chenyt9, 2023-11-24:
@Giantvince1 ,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-T1TH33.75-12-6-1-3

closed 2023-11-24 by chenyt9