MotorolaMobilityLLC/kernel-mtk #91: [Request] Kernel source code for motorola G54 5G

sarthakroy2002, 2023-09-29:
Please release kernel sources for motorola G54 5G.

asbraga-motorola, 2023-10-09:
Can you inform the exact released version (Build ID) you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2, T2TZ33.28-16-19.

Sureshgorai25, 2023-10-19:
Now the latest released version is T3TD33.16.51-1-1

sarthakroy2002, 2023-11-04:
@asbraga-motorola any updates on when it will be released!

sarthakroy2002, 2023-11-13:
@asbraga-motorola release!

sarthakroy2002, 2023-11-26:
@mbissaromoto @asbraga-motorola ping on this.. bump plz

sarthakroy2002, 2023-12-07:
@mbissaromoto @asbraga-motorola Please push!

chenyt9, 2023-12-13:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-T3TD33.16-22-5

closed 2023-12-13 by chenyt9