MotorolaMobilityLLC/kernel-mtk #104: LTS merge for Moto G73 5g aka devonf

rraanndd00mm, 2024-01-28:
Latest LTS kernel version is 5.10.209 as of 28/01/2024.. and moto g73 is on 5.10.168 and hence lacking critical kernel patches.. Please merge latest kernel security patches to this device's kernel.

asbraga-motorola, 2024-01-29:
Can you inform the exact released version (Build ID) you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2, T2TZ33.28-16-19.

rraanndd00mm, 2024-01-30:
No, this is not a request for kernel source.. Request was for merging the latest LTS kernel branch to this device's kernel.. Since the version T1TN33.14-90-9-3-1 has kernel 5.10.168 whereas the latest release is 5.10.209 so please update the kernel branch for the same.. Thankyou!

asbraga-motorola, 2024-01-30:
I understand, but this channel is basically for requesting kernel sources.
For that kind of request you may try Motorola support at https://www.motorola.com.br/ or Motorola Communities at https://forums.lenovo.com/t5/Motorola-Community/ct-p/MotorolaCommunity.

closed 2024-01-30 by asbraga-motorola