MotorolaMobilityLLC/kernel-mtk #113: [Request] Kernel source code for motorola G54 5G (U1TD34.94-12-7)Kernel Source for <build_id>

Sureshgorai25, 2024-04-09:
### Build ID

U1TD34.94-12-7

### Additional

Please release kernel sources for moto g54 5g android 14.
Build Number: U1TD34.94-12-7

asbraga-motorola, 2024-04-15:
Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-U1TD34.94-12-7

closed 2024-04-15 by asbraga-motorola

sarthakroy2002, 2024-04-20:
@asbraga-motorola Wlan and bt kernel modules U1TD34.94-12-7 aren't pushed 
Please push it

Sureshgorai25, 2024-04-21:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-U1TD34.94-12-7

Can you please push wlan/core/gen4m drivers of the same build

asbraga-motorola, 2024-05-09:
code for wlan/core/gen4m: https://github.com/MotorolaMobilityLLC/vendor-mediatek-kernel_modules-connectivity-wlan-core-gen4m/releases/tag/MMI-U1TD34.94-12-7

forforksake, 2024-11-09:
Can you update for U1TD34.94-12-9-10 please?

asbraga-motorola, 2024-11-11:
> Can you update for U1TD34.94-12-9-10 please?

Could you please open a new issue for that Build ID?