MotorolaMobilityLLC/kernel-mtk #124: [Request] Q Kernel source code for Lenovo Tab M10FHD Plus 2nd Gen (achilles6) QP1A.190711.020

jmpfbmx, 2024-07-20:
### Build ID

ro.build.id QP1A.190711.020
ro.build.display.id TB-X606X_S300673_220613_BMP

### Additional

Please release Q Kernel Source for Lenovo Tab M10FHD Plus 2nd Gen (achilles6) QP1A.190711.020 - TB-X606X_S300673_220613_BMP. @asbraga-motorola @chenyt9 @LONGCHEER-01 @mbissaromoto 

I am a developer which has this device and is interested on upstreaming kernel source at my LineageOS sources, etc.
My sources can be found here: @lenovo-mt6765

jmpfbmx, 2024-07-23:
Any update @asbraga-motorola @chenyt9 @LONGCHEER-01 @mbissaromoto ?

Bakoubak, 2024-08-09:
Could we get an update on this ? @asbraga-motorola @chenyt9 @LONGCHEER-01 @mbissaromoto ?

asbraga-motorola, 2024-08-09:
Hi,
These repos are for Motorola products, so we don't have access to information for Lenovo products. 

I did a quick web search and found this: https://support.lenovo.com/br/en/solutions/ht511330-lenovo-open-source-portal
Maybe it has what you are looking for.

Anyway, I'll add comments to this issue if I get more details.

closed 2024-08-09 by asbraga-motorola

Bakoubak, 2024-09-24:
> Hi, These repos are for Motorola products, so we don't have access to information for Lenovo products.
>
> I did a quick web search and found this: [...]

In theory it should have what we are looking for, but the sources there are broken and seems to cause weird bugs on our devices (broken ION allocator for X306X)