MotorolaMobilityLLC/kernel-mtk #130: Kernel Source for U1TDS34.94-12-7-6

sarthakroy2002, 2024-08-08:
### Build ID

U1TDS34.94-12-7-6

### Additional

OSS Kernel source release for moto g54 5G (cancunf) - U1TDS34.94-12-7-6

closed 2024-08-30 by asbraga-motorola

brinlyau, 2024-09-09:
Could you please also push the kernel modules for this release too?

asbraga-motorola, 2024-09-16:
You may look for the tag MMI-U1TDS34.94-12-7-6 in the modules repos, such as:

- motorola-kernel-modules
- vendor-mediatek-kernel_modules-connectivity-common
- vendor-mediatek-kernel_modules-connectivity-connfem
- vendor-mediatek-kernel_modules-connectivity-conninfra
- vendor-mediatek-kernel_modules-connectivity-fmradio
- vendor-mediatek-kernel_modules-connectivity-gps
- vendor-mediatek-kernel_modules-connectivity-wlan-core-gen4m
- vendor-mediatek-kernel_modules-gpu
- vendor-mediatek-kernel_modules-met_drv_v3
- ...

sarthakroy2002, 2024-09-16:
> You may look for the tag MMI-U1TDS34.94-12-7-6 in the modules repos, such as:
>
> * motorola-kernel-modules
> * vendor-mediatek-kernel_modules-connectivity-common
> * [...]

Can you please upload the kernel building guide on readme repo?