MotorolaMobilityLLC/kernel-mtk #136: Kernel source for Moto g23 Android 14

GitFASTBOOT, 2024-10-23:
### Build ID

UHAS34.29-3

### Additional

Providing kernel source can help us to take better experience

asbraga-motorola, 2024-11-08:
Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-UHAS34.29-3

closed 2024-11-08 by asbraga-motorola

GitFASTBOOT, 2025-01-26:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-UHAS34.29-3

Why you blocked us from unlock bootloader it's not fair moto g41 is lower price and unlockable please make moto g23 can be unlocked bootloader

R0rt1z2, 2025-03-28:
@asbraga-motorola would it be possible to also include penangf's `cust.dtsi` (`#include <penangf/cust.dtsi>`)?

asbraga-motorola, 2025-04-02:
> [@asbraga-motorola](https://github.com/asbraga-motorola) would it be possible to also include penangf's `cust.dtsi` (`#include <penangf/cust.dtsi>`)?

That file is a compiled product and is not included in the code, so we cannot provide it

R0rt1z2, 2025-04-02:
> > [@asbraga-motorola](https://github.com/asbraga-motorola) would it be possible to also include penangf's `cust.dtsi` (`#include <penangf/cust.dtsi>`)?
> 
> That file is a compiled product and is not included in the code, so we cannot provide it

@asbraga-motorola how about codegen.dws?

progzone122, 2025-08-07:
> [@asbraga-motorola](https://github.com/asbraga-motorola) how about codegen.dws?

@asbraga-motorola