MotorolaMobilityLLC/kernel-mtk #139: What toolchain should be used for building the Android kernel? (UHAS34.29-3 - moto g23)

progzone122, 2024-11-09:
# Build ID
UHAS34.29-3

# Logs
```bash
diablo-pc :: ~/Загрузки/kernel-mtk » make -j$(nproc --all) O=out ARCH=arm64 CC=$(pwd)/clang-llvm/bin/clang CROSS_COMPILE=aarch64-linux-gnu- CLANG_TRIPLE=aarch64-linux-gnu- LLVM_IAS=1
  GEN     Makefile
  CALL    ../scripts/atomic/check-atomics.sh
  CALL    ../scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  AR      drivers/firewire/built-in.a
  AR      drivers/cdrom/built-in.a
  AR      drivers/auxdisplay/built-in.a
  AR      drivers/macintosh/built-in.a
  AR      drivers/nvme/host/built-in.a
  CC      drivers/char/rpmb/rpmb-mtk.o
  CC      drivers/scsi/scsi_common.o
  CC      drivers/usb/phy/phy.o
  CC      drivers/usb/phy/of.o
  AR      drivers/net/phy/built-in.a
  CC      drivers/misc/mediatek/blocktag/blocktag-core.o
  CC      drivers/misc/utag/utag.o
  AR      drivers/nvme/target/built-in.a
  AR      drivers/nvme/built-in.a
  AR      drivers/net/mdio/built-in.a
  CC      drivers/usb/phy/phy-generic.o
  AR      drivers/net/pcs/built-in.a
  CC      drivers/gpu/mediatek/gpufreq/v2/gpufreq_mt6768.o
  CC      drivers/misc/mediatek/blocktag/blocktag-index.o
  AR      drivers/net/dsa/b53/built-in.a
  AR      drivers/net/dsa/microchip/built-in.a
  CC      drivers/misc/mediatek/cm_mgr/mtk_cm_mgr_common.o
  AR      drivers/net/dsa/mv88e6xxx/built-in.a
  AR      drivers/net/dsa/ocelot/built-in.a
  CC      drivers/gpu/mediatek/ged/src/ged_kpi.o
  AR      drivers/net/dsa/qca/built-in.a
  AR      drivers/net/dsa/sja1105/built-in.a
  AR      drivers/net/dsa/built-in.a
  AR      drivers/net/built-in.a
  AR      drivers/scsi/built-in.a
../drivers/misc/mediatek/qos/Makefile:33: extraneous text after 'else' directive
  CC      drivers/misc/mediatek/qos/mtk_qos_common.o
  CC      drivers/misc/mediatek/ccci_util/ccci_util_lib_fo.o
../drivers/gpu/mediatek/ged/src/ged_kpi.c:54:10: fatal error: 'disp_arr.h' file not found
   54 | #include "disp_arr.h"
      |          ^~~~~~~~~~~~
1 error generated.
make[5]: *** [../scripts/Makefile.build:273: drivers/gpu/mediatek/ged/src/ged_kpi.o] Error 1
make[4]: *** [../scripts/Makefile.build:516: drivers/gpu/mediatek/ged] Error 2
make[4]: *** Waiting for unfinished jobs....
  CC      drivers/misc/mediatek/blocktag/blocktag-ufs.o
  AR      drivers/usb/typec/altmodes/built-in.a
  AR      drivers/usb/typec/mux/built-in.a
  CC      drivers/misc/mediatek/qos/mtk_qos_bound.o
  CC      drivers/usb/typec/class.o
../drivers/gpu/mediatek/gpufreq/v2/gpufreq_mt6768.c:2688:3: error: ignoring return value of function declared with 'warn_unused_result' attribute [-Werror,-Wunused-result]
 2688 |                 PTR_ERR(efuse_cell);
      |                 ^~~~~~~ ~~~~~~~~~~
../drivers/gpu/mediatek/gpufreq/v2/gpufreq_mt6768.c:2695:6: error: ignoring return value of function declared with 'warn_unused_result' attribute [-Werror,-Wunused-result]
 2695 |             PTR_ERR(efuse_buf);
      |             ^~~~~~~ ~~~~~~~~~
  CC      drivers/misc/mediatek/qos/mtk_qos_sysfs.o
2 errors generated.
make[6]: *** [../scripts/Makefile.build:273: drivers/gpu/mediatek/gpufreq/v2/gpufreq_mt6768.o] Error 1
make[5]: *** [../scripts/Makefile.build:516: drivers/gpu/mediatek/gpufreq/v2] Error 2
make[4]: *** [../scripts/Makefile.build:516: drivers/gpu/mediatek/gpufreq] Error 2
make[3]: *** [../scripts/Makefile.build:516: drivers/gpu/mediatek] Error 2
make[2]: *** [../scripts/Makefile.build:516: drivers/gpu] Error 2
make[2]: *** Waiting for unfinished jobs....
In file included from ../drivers/misc/mediatek/blocktag/blocktag-index.c:25:
[... 71 lines cut ...]
  AR      drivers/usb/built-in.a
  CC      drivers/misc/mediatek/ccci_util/ccci_util_md_rat.o
make[2]: *** [../scripts/Makefile.build:516: drivers/char] Error 2
  AR      drivers/misc/mediatek/ccci_util/built-in.a
make[3]: *** [../scripts/Makefile.build:516: drivers/misc/mediatek] Error 2
make[2]: *** [../scripts/Makefile.build:516: drivers/misc] Error 2
make[1]: *** [/home/diablo/Загрузки/kernel-mtk/Makefile:1921: drivers] Error 2
make: *** [Makefile:192: __sub-make] Error 2
```

```bash
diablo-pc :: ~/Загрузки/kernel-mtk » bazel build --config=fast //common:kernel_aarch64_dist
WARNING: Invoking Bazel in batch mode since it is not invoked from within a workspace (below a directory
having a WORKSPACE file).
OpenJDK 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will
likely be removed in a future release.
ERROR: The 'build' command is only supported from within a workspace (below a directory having a WORKSPAC
E file).
See documentation at https://bazel.build/concepts/build-ref#workspace
```

```bash
diablo-pc :: ~/Загрузки/kernel-mtk 2 » bash build/build.sh
bash: build/build.sh: No such file or directory
```

cyberknight777, 2024-11-12:
Hi @hq-zxm, I know this is unrelated to this github issue but could you check out https://github.com/MotorolaMobilityLLC/vendor-mediatek-kernel_modules-gpu/issues/3. Additionally all vendor/mediatek drivers for moto g54 - U1TDS34.94-12-7 are wrong drivers. They are for mediatek 6.1 kernel devices. The drivers that you have pushed for moto g23 - UHAS34.29-3 are the correct ones for moto g54. Please fix this by uploading the correct source on that branch, thanks.

closed 2024-12-04 by asbraga-motorola

asbraga-motorola, 2025-01-20:
The label MMI-U1TDS34.94-12-7-6 for the drivers is fixed.

Do not forget to delete your local tags before fetching the new one: git tag -d MMI-U1TDS34.94-12-7-6.