MotorolaMobilityLLC/kernel-mtk #145: Kernel Source for VVTA35.51-28-24 (G15 / lamu)

zainarbani, 2025-01-20:
### Build ID

VVTA35.51-28-24

### Build fingerprint

motorola/lamu_g/lamu:15/VVTA35.51-28-24/f23912:user/release-keys

shomykohai, 2025-09-21:
@asbraga-motorola I was able to download the published kernel (even though zip files were published again...), though I don't see any build instructions. Can you add them to the README repo?

TinnoOSS, 2025-09-25:
Hi @shomykohai 
python $KERNEL_DIR/scripts/gen_build_config.py --kernel-defconfig lamu_defconfig -m user -o $my_top_dir/out/target/product/lamu/obj/KERNEL_OBJ/build.config
 
Comment this line; it will not be included in the build.

TinnoOSS, 2025-09-26:
Hi @TinnoOSS 
Some of the missing repositories can be downloaded through branch: android-15-release vta35, but due to some repositories not being open source, some repositories cannot be downloaded, so there may still be errors. May I ask what you want to do? If it is for successful compilation, we will ask our engineers to provide a new compilation script.

shomykohai, 2025-09-26:
> Hi @TinnoOSS
> Some of the missing repositories can be downloaded through branch: android-15-release vta35, but due to some repositories not being open source, some repositories cannot be downloaded, [...]

Hi, my goal is to be able to compile a working and bootable kernel.
Thanks for the hint for some of those modules, I'll check which one require this branch.

shomykohai, 2025-09-28:
@TinnoOSS still no luck in compiling the kernel. Is it possible to get a new compilation script?

shomykohai, 2025-10-22:
@TinnoOSS @asbraga-motorola I still can't build the kernel. I get several errors, and lamu_defconfig doesn't even exists under `arch/arm64/configs`

Can you please provide a new build script?

R0rt1z2, 2025-11-03:
Hello @TinnoOSS @asbraga-motorola any news regarding this?

bengris32, 2025-11-11:
@TinnoOSS @asbraga-motorola Hi, any updates on the release?

artempeshkov, 2025-11-18:
> [@TinnoOSS](https://github.com/TinnoOSS) It has been nearly a month since your last response. Could you please share any updates or progress?

no new branch appeared in android-15-release-v1tl35.73-60-3 for vvta35

artempeshkov, 2025-11-18:
im looking for clean 6769 6.6 sources too

TinnoOSS, 2025-11-20:
@bengris32 The released kernel repository is the content of kernel/kernel device modules 6.6, while the rest are compilation tools and environments. The required compilation tools and environments have been noted in the shell script

TinnoOSS, 2025-12-06:
You can refer to this readme. Please ask if you have any questions.
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-VVTA35.51-28-24.txt

bengris32, 2025-12-07:
> You can refer to this readme. Please ask if you have any questions. https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-VVTA35.51-28-24.txt

So are you going to upload the actual `kernel_device_modules-6.6` repo instead of telling us to rename `kernel-6.6` to `kernel_device_modules-6.6`? Those modules are still under GPL and nothing you can do is going to change that ;)

halal-beef, 2025-12-07:
> You can refer to this readme. Please ask if you have any questions. https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-VVTA35.51-28-24.txt

Lad this says Lagos

TinnoOSS, 2025-12-07:
Sorry, I uploaded the wrong readme file. I've re-uploaded it. Thank you for reminding me.

bengris32, 2025-12-07:
> Sorry, I uploaded the wrong readme file. I've re-uploaded it. Thank you for reminding me.

Hiiiiiiiiiiiiiiiiiiii

Please upload the `kernel_device_modules-6.6` repo to https://github.com/MotorolaMobilityLLC/kernel-kernel_device_modules-6.6
As you can see from the existing source drops for different devices (**from compliant ODMs unlike yours**), code under that path is GPL 2.0!

[image]

halal-beef, 2025-12-07:
[image]

So as you probably know, you should release the kernel modules in general

erorndev, 2025-12-07:
BlackRock is the world's largest asset manager, overseeing around $10 trillion in assets. Its basically Jews again, and this issue correlates to ashkenazi entities.

progzone122, 2025-12-07:
@TinnoOSS literally

[image]

halal-beef, 2025-12-07:
@TinnoOSS Please upload the kernel modules 😊 THANKS!

[image]

Wqttzicue, 2025-12-07:
[image]

InternalHellhound, 2025-12-07:
My father @TinnoOSS works at Motorola. Today he got an urgent call to a meeting because of GPLv2 violation. He came back late and didn’t explain anything. He just said to pack our things and run to the store for two weeks’ worth of supplies. Now we're driving somewhere far outside the city. I don’t know what’s going on, but it feels like it’s starting...

halal-beef, 2025-12-07:
[image]

Hello from mentally sane mainliners

TheFreo, 2025-12-07:
[image] Hi from Nakama

Sil-fait-mourir, 2025-12-07:
You're valodating GPLv2 . Read Aosp licences.

erorndev, 2025-12-07:
[image]

who wants to play Roblox with me?

InternalHellhound, 2025-12-07:
[image]

Заходит мужик на кухню, а там Армяне с @TinnoOSS не выкладывают модули ядра

Sil-fait-mourir, 2025-12-07:
https://www.gnu.org/licenses/gpl-violation.html

erorndev, 2025-12-07:
There are several compelling reasons not to violate the GPL (GNU General Public License):
Legal consequences:
The GPL is a legally binding contract. Violating it means you lose your license to use, modify, and distribute the software
Copyright holders can sue for infringement, potentially leading to injunctions, financial damages, and legal fees
Organizations like the Software Freedom Conservancy and FSF actively enforce GPL compliance.

roxysqq, 2025-12-07:
[image]

# I use Arch, btw

karkar47, 2025-12-07:
> <img alt="Image" width="1280" height="852" src="https://private-user-images.githubusercontent.com/242938559/523500098-1f954d65-2a28-47ef-864b-2f9b506161eb.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjUxMzc3MjMsIm5iZiI6MTc2NTEzNzQyMywicGF0aCI6Ii8yNDI5Mzg1NTkvNTIzNTAwMDk4LTFmOTU0ZDY1LTJhMjgtNDdlZi04NjRiLTJmOWI1MDYxNjFlYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTIwN1QxOTU3MDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OWRiYzcyZGVjMTAwOWM3NGU2ZmUyZTZiMTY4YTUwNDk0YTdhM2UyOGJkY2E0ZWFkMmM0Y2MzM2NiYTFkNDZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.NgCU-9s5AGD0fTeyB0V7GTTKPAwDXPbpiGulLFoWX5A">
> # I use Arch, btw

Диабло доволен Вами.

erorndev, 2025-12-07:
> > There are several compelling reasons not to violate the GPL (GNU General Public License): Legal consequences: The GPL is a legally binding contract. Violating it means you lose your license to use, modify, and distribute the software Copyright holders can sue for infringement, potentially leading to injunctions, financial damages, and legal fees Organizations like the Software Freedom Conservancy and FSF actively enforce GPL compliance.
> 
> cool joke

[image]

volrod64, 2025-12-07:
[image] 
Shame.

fenolftalein-mamir, 2025-12-07:
chat just do sudo apt install kernel_device_modules 😭

0vercoin, 2025-12-07:
> chat just do sudo apt install kernel_device_modules 😭

[image]

erorndev, 2025-12-07:
[image]

> > chat just do sudo apt install kernel_device_modules 😭
> 
> [image]

fenolftalein-mamir, 2025-12-07:
> > chat just do sudo apt install kernel_device_modules 😭
> 
> <img alt="Image" width="739" height="688" src="https://private-user-images.githubusercontent.com/166366513/523501535-08a2e362-2844-4e26-b173-fff830754d44.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjUxMzg4ODYsIm5iZiI6MTc2NTEzODU4NiwicGF0aCI6Ii8xNjYzNjY1MTMvNTIzNTAxNTM1LTA4YTJlMzYyLTI4NDQtNGUyNi1iMTczLWZmZjgzMDc1NGQ0NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTIwN1QyMDE2MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMTkwMGY1ZGE1NGFjNzcxNjU0NjA5OGNiZThjZWM4NmIwMmVkYWNhMWI2MTY3N2QwMjUzMDgxZmM3YTVkYWVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.grwIYQXpuYsSGDYZBuSCsPCuAI_O7n6lGG_hUQZTl4w">

i dont comply with your resisting of laughing

progzone122, 2025-12-07:
@TinnoOSS 

If you do not provide us with a working kernel, modules, and everything else needed for a successful kernel build, we will have to contact [SFC](https://sfconservancy.org/).
You don't want that, do you? We would also like to avoid unnecessary scandals and wasting time.

We await your further actions.

gentoocat, 2025-12-07:
who cares about gpl2 in China

erorndev, 2025-12-07:
> [image]
> 
> I'm back, chat

it was boring without you lowk

harrywhittick, 2025-12-07:
me when the break law gpl

TheMorc, 2025-12-07:
> who cares about gpl2 in China

mačk

erorndev, 2025-12-07:
> >mačk
> 
> Huh 

c a t

harrywhittick, 2025-12-07:
> maćk
>> Huh
>>> c a t

I eat dogs

gentoocat, 2025-12-07:
motorola, use harmonyos
its better than aosp and has closed kernel source

Daniel224455, 2025-12-07:
[image]

AndreiSva, 2025-12-07:
A lot of unprofessional discourse in this thread it seems.

ravenekse, 2025-12-07:
[image]

hoovad, 2025-12-07:
> A lot of unprofessional discourse in this thread it seems.

your mum

InternalHellhound, 2025-12-07:
Billions must aniship

Daniel224455, 2025-12-07:
Billions must forget MediaTek exploiting and move on to Qualcomm

erorndev, 2025-12-07:
billions must rate my cat
[image]

Toquita3D, 2025-12-07:
Hey @torvalds you might wanna check this one.

User8395, 2025-12-07:
bro pinged The Man Himself 🥀

erorndev, 2025-12-07:
> Hey @torvalds you might wanna check this one.

[image]

zyuzyechka, 2025-12-07:
@torvalds 

[image]

https://github.com/user-attachments/assets/48aaaf6b-841c-4122-8be7-15693f8f6228
https://github.com/user-attachments/assets/567ac05f-9c58-42c5-8e61-7f320a79ce75
https://github.com/user-attachments/assets/50ae110b-0717-4412-a107-832a7c6262bb
https://github.com/user-attachments/assets/cca0593a-9186-4b0f-953a-d846bc53cdd8

> Hey [@torvalds](https://github.com/torvalds) you might wanna check this one.

bomb

progzone122, 2025-12-07:
> > Hey [@torvalds](https://github.com/torvalds) you might wanna check this one.
> 

[image]

Daniel224455, 2025-12-07:
> > > [image]
> > > This chat is on fire. Just like someone who can be fired from their job if they continue violating the GPL v2 license. Nobody would want that 😉
> > 
> > 
> > ohmygod WHY ARE YOU HERE
> 
> Are gathering entire silicium here atp? 😭😭😭

ye

renhiyama, 2025-12-07:
> > A lot of unprofessional discourse in this thread it seems.
> 
> I think the drawings here were highly professional.

The only one being unprofessional is Tinno by violating GPL v2 license 🥀

renhiyama, 2025-12-07:
[image]

[15 more replies]