MotorolaMobilityLLC/kernel-mtk #148: Kernel Source for UTA34.82-97

GitFASTBOOT, 2025-02-24:
### Build ID

UTA34.82-97

### Build fingerprint

motorola/fogorow_g/fogorow:14/UTA34.82-97/82-97:user/release-keys

### Additional

We need kernel source and modules for moto g24 pls

progzone122, 2025-02-25:
Yes, please provide us with the sources

R0rt1z2, 2025-03-24:
bump. @asbraga-motorola

R0rt1z2, 2025-03-26:
Maybe @TinnoOSS?

progzone122, 2025-03-27:
@asbraga-motorola @TinnoOSS

shomykohai, 2025-03-27:
@TinnoOSS @asbraga-motorola

asbraga-motorola, 2025-03-28:
Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-UTA34.82-97

closed 2025-03-28 by asbraga-motorola

R0rt1z2, 2025-03-28:
@asbraga-motorola @TinnoOSS are you sure these are correct? I can’t find the defconfig nor the DTS. I think you’ve released the wrong sources, can you please recheck?

TinnoOSS, 2025-03-31:
@R0rt1z2 The latest code has been merged. Enclosed please find

R0rt1z2, 2025-03-31:
> [@R0rt1z2](https://github.com/R0rt1z2) The latest code has been merged. Enclosed please find

Thanks. What defconfig should we use? I'm still not sure.

R0rt1z2, 2025-03-31:
> [@R0rt1z2](https://github.com/R0rt1z2) The latest code has been merged. Enclosed please find

@TinnoOSS @asbraga-motorola Thank you. What defconfig should we use? I'm a bit confused. Are there build instructions available?

TinnoOSS, 2025-04-01:
@R0rt1z2 I have uploaded a README.txt file, please refer to this file

asbraga-motorola, 2025-04-01:
The branch `android-14-release-UTA34.82-97` was incorrectly merged to the branch `android-13-release-ttt` which has no relation to this product.
I fixed it by reseting `android-13-release-ttt` to its original commit and repositioning the tag MMI-UTA34.82-97 to the HEAD of `android-14-release-UTA34.82-97`.

Please, remove the old local branch or local tag you may have and fetch the repo again.