MotorolaMobilityLLC/kernel-mtk #154: Kernel Source for V1TDS35H.83-20-5-2

cyberknight777, 2025-04-01:
### Build ID

V1TDS35H.83-20-5-2

### Build fingerprint

motorola/cancunf_g_hal/cancunf:12/V1TDS35H.83-20-5-2/ad0dc:user/release-keys

asbraga-motorola, 2025-04-08:
Code published: https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-V1TDS35H.83-20-5-2

cyberknight777, 2025-04-09:
@asbraga-motorola where are the sources for https://github.com/MotorolaMobilityLLC/vendor-mediatek-kernel_modules-connectivity-bt-mt66xx and https://github.com/MotorolaMobilityLLC/vendor-mediatek-kernel_modules-connectivity-wlan-adaptor ?

Moto G54 5G (cancunf) uses those modules, so kindly push sources for them, thanks!

cyberknight777, 2025-04-09:
Moto G54 5G (cancunf) uses these:

kernel-mtk 
vendor-mediatek-kernel_modules-connectivity-bt-mt66xx
vendor-mediatek-kernel_modules-connectivity-common
vendor-mediatek-kernel_modules-connectivity-connfem
vendor-mediatek-kernel_modules-connectivity-fmradio
vendor-mediatek-kernel_modules-connectivity-gps
vendor-mediatek-kernel_modules-connectivity-wlan-adaptor
vendor-mediatek-kernel_modules-connectivity-wlan-core-gen4m
vendor-mediatek-kernel_modules-met_drv_v3
vendor-mediatek-kernel_modules-gpu
motorola-kernel-modules

Out of all this, what’s missing is met_drv_v3, bt-mt66xx and wlan-adaptor. @asbraga-motorola please push them, thanks.