MotorolaMobilityLLC/kernel-mtk #162: Kernel Source for Motorola G Pure (ellis) s3rhs32

PolysheepCodes, 2025-07-03:
### Build ID

S3RHS32.20-42-10-4-11

### Build fingerprint

s3rhs32

### Additional

i would like to request a more inclusive dump of the kernel source for mt6765 based devices as the current available build appears to be inconclusive and missing important header files required for power management and building the device kernel

PolysheepCodes, 2025-07-04:
> Can you provide more details about your problem? What headers are missing?

My initial post was apparently wrong, upon further inspection I have all the headers I need but I'm also having weird issues I can't seem to solve with make parameters or editing the KConfig specifically involving that header issue I was having because I had to manually hunt down every file and move it to the needed folders but I'm also having a weird issue I can't even figure out, I will share my compiler logs later when I can, but for now could you help me troubleshoot having to move stuff manually and why I can't figure out how to define MTK_PLATFORM in the make command?

ETJAKEOC, 2025-08-07:
I too am curious about the chances of getting the kernel source for this device, and the required blobs to work on compiling my own OS for the device. I'm no stranger to compiling kernels, I currently ship 24 pre-built kernels for 6 different Linux distributions, both GCC, and LLVM builds, but haven't really stepped into the ARM space yet. But as this device is far beyond EOL, I would like to hope in good faith that Motorola would release these sources to us users who want to maintain the devices outselves.

PolysheepCodes, 2025-08-07:
> I too am curious about the chances of getting the kernel source for this device, and the required blobs to work on compiling my own OS for the device. I'm no stranger to compiling kernels, I [...]

so I have an update on this, I do have the most up to date kernel compatible with the Motorola Ellis, I too am working on a port of Linux for the device and getting closer but having some issues regarding device blobs, to find the kernel search up s3rhs32 in the branches menu and it'll be the only result, but I'd like to ask if I could get in touch with you so I could try and solve my issue

ETJAKEOC, 2025-08-07:
> so I have an update on this, I do have the most up to date kernel compatible with the Motorola Ellis, I too am working on a port of Linux for the device and getting closer but having some issues [...]

Sure thing, if you got Discord, you can find me there under the same username. Now you piqued my interest with a Linux port though 😂

closed 2025-08-15 by asbraga-motorola